synthread/Klippventurer #3: Full Klipper ROM config?

dJOS1475, 2022-08-07:
Howdy, I finally got around to removing the old MIPS MCU (good riddance) and have successfully flashed STM32Duino.bin and my own klipper.bin to the specified ROM locations. However, I'm not getting a successful startup. Can you please confirm the full Klipper Firmware Configuration options please?

                         Klipper Firmware Configuration
[*] Enable extra low-level configuration options
    Micro-controller Architecture (STMicroelectronics STM32)  --->
    Processor model (STM32F103)  --->
[ ] Only 10KiB of RAM (for rare stm32f103x6 variant)
[ ] Disable SWD at startup (for GigaDevice stm32f103 clones)
    Bootloader offset (28KiB bootloader)  --->
    Clock Reference (12 MHz crystal)  --->
    Communication interface (Serial (on USART1 PA10/PA9))  --->
(250000) Baud rate for serial port
()  GPIO pins to set at micro-controller startup

VioSynthax, 2022-08-07:
Hello, 

Try using the included Klipper bin for your initial setup to make sure everything else in your setup is working and your MCU is able to be communicated with from your CPU (Pi or otherwise). If Klipper needs the MCU (STM32 or clone) to be updated, then go ahead and compile the newer controller firmware for it. Those options don’t look right, but it’s been a while so I’ll need to take a look at the files and see what the config should be.

dJOS1475, 2022-08-07:
no worries, It's always nice to know the settings just in case you need to recompile (happens sometimes with klipper updates)

dJOS1475, 2022-08-07:
Hmm, even with your FW I just get what sounds like a boot loop with lots of clicking noises from the little speaker. Do I need the DFU jumper connected for the klipper.bin? I had it there for the stm32 bin and STM32 cube programmer has been reporting successful programs. eg

   21:31:50 : ST-LINK SN  : 490046000100004D32414D4E
  21:31:50 : ST-LINK FW  : V2J29S7
  21:31:50 : Board       : --
  21:31:50 : Voltage     : 3.19V
  21:31:50 : SWD freq    : 4000 KHz
  21:31:50 : Connect mode: Normal
  21:31:50 : Reset mode  : Software reset
  21:31:50 : Device ID   : 0x414
  21:31:50 : Revision ID : --
  21:31:50 : Debug in Low Power mode is not supported for this device.
  21:31:50 : UPLOADING OPTION BYTES DATA ...
  21:31:50 :   Bank          : 0x00
  21:31:50 :   Address       : 0x4002201c
  21:31:50 :   Size          : 8 Bytes
  21:31:50 :   Bank          : 0x01
  21:31:50 :   Address       : 0x1ffff800
  21:31:50 :   Size          : 16 Bytes
  21:31:50 : UPLOADING ...
  21:31:50 :   Size          : 1024 Bytes
  21:31:50 :   Address       : 0x8000000
  21:31:50 : Read progress:
  21:31:50 : Data read successfully
  21:31:50 : Time elapsed during the read operation is: 00:00:00.006
  21:31:53 : Memory Programming ...
  21:31:53 : Opening and parsing file: klipper.bin
  21:31:53 :   File          : klipper.bin
  21:31:53 :   Size          : 21.06 KB 
  21:31:53 :   Address       : 0x08002000 
  21:31:53 : Erasing memory corresponding to segment 0:
  21:31:53 : Erasing internal memory sectors [4 14]
  21:31:54 : Download in Progress:
  21:31:55 : File download complete
  21:31:55 : Time elapsed during download operation: 00:00:01.757
  21:31:55 : Verifying ...
  21:31:55 : Read progress:
  21:31:55 : Download verified successfully 
  21:32:02 : Disconnected from device.

VioSynthax, 2022-08-07:
> Hmm, even with your FW I just get what sounds like a boot loop with lots of clicking noises from the little speaker. Do I need the DFU jumper connected for the klipper.bin? I had it there for the [...]

Hmm, looks like it might be flashed to the wrong location, there's an extra 0 in the address you're flashing the klipper.bin firmware to. Reflash your bootloader to make sure it's intact at 0x08000000, then flash klipper.bin at 0x8002000. I seem to recall an issue where my STM32 wasn't booting, with the solution being to use these addresses exactly this way. You can also see what error you're getting stopping it from booting by using the fault analyzer in CubeProg. As far as I can tell, both the STM32 used on this board and its clones use the 0x08 as their boot address, and the bootloader should be using the 0x8002 as the location it looks to launch firmware from. I'd copy paste those as-is and see if anything changes.

dJOS1475, 2022-08-07:
Ok, I tried that and went through the whole process again and same result. Do I need the DFU jumper everytime, or was that just to unlock the eeprom the first time?

Here are my programmer settings just in case im doing something wrong (klipper.bin):
https://imgur.com/a/tu92wnX

VioSynthax, 2022-08-07:
> Ok, I tried that and went through the whole process again and same result. Do I need the DFU jumper everytime, or was that just to unlock the eeprom the first time?
> 
> Here are my programmer settings just in case im doing something wrong (klipper.bin): https://i.imgur.com/fzPtGNJ.jpg

Set to skip erase before flashing or you'll erase your bootloader while writing klipper. Also, I think your start address on the CPU page needs to remain 0x08000000 so it knows where the bootloader's first instruction is, but you'll need to set the flash address to 0x8002000 while flashing klipper. Been a while since I've had time to work on my printer, but I'll try to make sure you get yours up and running. Iirc you'll need to jump the DFU pins on startup, then flash bootloader (and not checking boot after flashing, make sure you aren't wiping the flash while you program firmware to the 2nd address. Then you can power off and see if it's being detected over the Pi's serial port (are you using a Pi? Which one? There's a process to set up the serial port on them so it doesn't get reserved for the Bluetooth module, for the Pis that have one)

VioSynthax, 2022-08-07:
My Discord is Synthax#0001 if you want some more real-time assistance, just remind me to post or post the solution yourself here so I can close the issue :)

dJOS1475, 2022-08-07:
Cheers, Im a noob when it comes to STM32 flashing. If I skip flash erase, it fails to download file. However when I tried this it worked:

https://i.imgur.com/pOsXdWY.jpg

but I still get the boot loop and clicking. Anyway dude, thanks for the help so far, Im off to bed. I'll have another swing at it tomorrow.

dJOS1475, 2022-08-07:
btw, Im using a Pi3A with Mainsail installed via the custom image from github

dJOS1475, 2022-08-08:
Argh, I can program the STM32 to my hearts content, and can read back both address locations without issue and compare between programmed and erased. But I can just cant connect to the dam thing from my Pi.

I solved the clicking / reboot issue - that was my fault. I got my serial wires mixed up and had GND and TX flipped. 🤦‍♂️

dJOS1475, 2022-08-08:
Hmmm, after a bit more digging, I have "GD32F103 RCT6" with 10KiB of RAM and 256KiB of Flash made by GigaDevice. Im wondering if that is why it wont play nice?

https://gd32mcu.com/data/documents/datasheet/GD32F103xx_Datasheet_Rev2.12.pdf

dJOS1475, 2022-08-08:
So I tried a different Pi3B with no change. I also tried changing the UART pins when compiling the Klipper binary and I got an leg to turn on in the heater section of the board.

I’ve also been able to confirm that the flash is being written currently by doing a file to rom compare in the STM32 tool. So that confirms that side of things is good.

i still can’t connect to the blasted UART port tho and I’m wondering if you have a real STM32 on your board? Mine is the GD32 which I’ve found info suggesting it isn’t 100% identical to the real thing.

dJOS1475, 2022-08-09:
Ok, I've tried connecting to the board with Pronterface via serial and I cant get it to respond either - I may have killed the serial interface. :-( But it's be great If you can confirm which STM32 you have? 

I've tried your Klipper ROM and also a made ROM's using every UART option with now result.

VioSynthax, 2023-03-14:
Version 2.0 of the guide improves the process and eliminates these issues. Should be pretty straightforward now!

closed 2023-03-14 by VioSynthax

VioSynthax, 2023-03-14:
Still in the 2.0-preview branch as the software setup part instructions and script aren't ready, but it's not too complicated. Gotta enable your UART port and disable bluetooth on the Pi, run KIAUH, copyprinter.cfg where Klipper wants it (can use Fluidd or Mainsail for that), and `make` klipper.bin from ~/klipper with my .config here: https://github.com/VioSynthax/Klippventurer-Installer/blob/main/configs/adventurer3.config

then do `sudo apt install stm32flash` followed by `sudo stm32flash -w ~/klipper.bin -R -i -18,23,18:-18,-23,18 /dev/ttyAMA0`
Keep in mind that command will only work if the Pi is connected as it is in v2 of the guide- https://github.com/VioSynthax/Klippventurer/tree/2.0-preview