synthread/Klippventurer #5: TMC2208's?

ldw300, 2023-02-21:
Is it a problem that the part number on the driver chips of my mainboard is TMC2208-LA? The config and README says TMC2209's. I have an adventurer 3 pro. Seems to be the same mainboard as in the picture.

VioSynthax, 2023-02-21:
> Is it a problem that the part number on the driver chips of my mainboard is TMC2208-LA? The config and README says TMC2209's. I have an adventurer 3 pro. Seems to be the same mainboard as in the picture.

Simple to switch the driver version, just changing all instances of 2209 in config to 2208. You can do a "replace all" for 2209>2208. If you go for it, please report back about your results and any issues you run into along the way! Happy to help in any way I'm able.

ldw300, 2023-02-24:
Gotcha, I'm also having issues connecting to the board to flash it, I have the mediatek chip off and jumper wires soldered in place, but I've tried a usb to ttl and an ftdi cable to connect and i just get ERROR: Activating Device KO from the cube programmer. I can open a seperate issue if needed.

VioSynthax, 2023-02-24:
You may have to use an ST-Link (or clone) like this one https://www.amazon.com/gp/product/B07SQV6VLZ

ldw300, 2023-02-26:
I looked at the clone but was unsure about them. I haven't had good luck with serial communication on other projects as is. So I just ordered the ST-Link v2 from amazon. Little pricey but I think it should work for other future projects.

VioSynthax, 2023-03-15:
ST-Link is no longer required, the new guide uses the Pi to do the flashing and even automates the DFU and reset process. You can return that ST-Link V2 and get your $60 back :)

ldw300, 2023-03-15:
Sounds great!

VioSynthax, 2024-02-28:
TMC 2208 is now the default config. Adventurer 3 Pro will have a separate config for TMC 2209.

closed 2024-02-28 by VioSynthax