synthread/Klippventurer #6: Install without removing MediaTek board

VioSynthax, 2023-02-25:
It should be possible to, in lieu of removing the entire daughter board assembly, prevent it from booting by soldering a jumper between its reset pin and ground, or by bridging its oscillator to ground. Then the normal jumper wires mentioned in the guide can be soldered with the MT CPU board still in place, fully disabling and bypassing it in a way that allows for the printer to be returned to stock unmodified by just removing the jumpers and installing the FF Finder firmware back onto the STM32. 

Please reply below if you are interested in trying this method, I'll create a branch soon with the simplified instructions once one or more people show interest.

j-collom, 2023-02-25:
I'm very interested in this, can you please send more information?

ldw300, 2023-02-26:
I am interested, I could not get my UART serial cables to connect so I resoldered the mediatek chip back onto the mainboard and am using the printer now. At least I was waiting for the st-link I ordered to come in before trying again.

VioSynthax, 2023-02-26:
Working on new instructions that should allow this to work, only thing remaining to do is locate the reset point on the daughter board.

VioSynthax, 2023-03-11:
Alright, update time. I've soldered the MediaTek module "MT" back onto the mainboard. I'm going to completely clear my STM32 memory and start over, so I can try to reproduce the issue people are having with getting the bootloader to jump to the firmware. I've made some many changes to the guide (currently not in the main branch) that should help someone new to modding accomplish this mod! I have some new features planned as well, I hope with my loaded schedule I have time to implement them, they'd really make this printer into a really solid machine.

**I need some community help here**- could one of you hook up your STLink to their stock board and use CubeProg to read the contents of their flash and send it to me? (dropdown arrow beside "Read" > Save As ...) I'm hoping to add a "revert to stock" option, which along with the MT disable mod in lieu of a full MT delete should officially make this what you could call a "no cut" mod!

penz6, 2023-04-24:
I am also very interested in it, have an Adventurer 3 that would be awesome with klipper, not very proficient in soldering but feel confident with the software portion.

hw-lunemann, 2023-04-29:
I am currently working on porting latest OpenWRT onto the adv3 mainboard.
```
BusyBox v1.36.0 (2023-04-29 10:58:48 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt SNAPSHOT, r22665-626c84340d
 -----------------------------------------------------
=== WARNING! =====================================
There is no root password defined on this device!
Use the "passwd" command to set up a new password
in order to prevent unauthorized SSH logins.
--------------------------------------------------
root@OpenWrt:~# uname -a
Linux OpenWrt 5.15.108 #0 Sat Apr 29 10:58:48 2023 mips GNU/Linux
```

As far as I can tell so far all of the hardware should be working already.
What's left is figuring out the logistics of getting this installed easily.
Rooting the device is quite simple since the firmware updates run unprotected bash scripts as root and
from there it's going to be possible to install my port of OpenWRT.  With the way flashforge set up the Linux
install it should even easily be possible to install it side by side with the original operating system.

With that setup it should also not be too hard to port the Klipper host software onto the machine and eliminate the need for a raspberry pi entirely. Flashing and communicating with the stm32 daughterboard can be done over UART (ttyS1).

Also both the screen and the touchscreen are controlled over spi so we could even get that functionality back but with Klipper! 

Bonus points: This can all be delivered by hijacking the firmware update process of the adv3 by simply plugging in a USB.

VioSynthax, 2023-04-29:
> I am currently working on porting latest OpenWRT onto the adv3 mainboard.
> 
> ```
> BusyBox v1.36.0 (2023-04-29 10:58:48 UTC) built-in shell (ash)
> 
>   _______                     ________        __
>  |       |.-----.-----.-----.|  |  |  |.----.|  |_
>  |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
>  |_______||   __|_____|__|__||________||__|  |____|
>           |__| W I R E L E S S   F R E E D O M
>  -----------------------------------------------------
>  OpenWrt SNAPSHOT, r22665-626c84340d
>  -----------------------------------------------------
> === WARNING! =====================================
> There is no root password defined on this device!
> Use the "passwd" command to set up a new password
> in order to prevent unauthorized SSH logins.
> --------------------------------------------------
> root@OpenWrt:~# uname -a
> Linux OpenWrt 5.15.108 #0 Sat Apr 29 10:58:48 2023 mips GNU/Linux
> ```
> 
> As far as I can tell so far all of the hardware should be working already. What's left is figuring out the logistics of getting this installed easily. Rooting the device is quite simple since the firmware updates run unprotected ash scripts as root and from there it's going to be possible to install my port of OpenWRT. With the way flashforge set up the Linux install it should even easily be possible to install it side by side with the original operating system.
> 
> With that setup it should also not be too hard to port the Klipper host software onto the machine and eliminate the need for a raspberry pi entirely. Flashing and communicating with the stm32 daughterboard can be done over UART (ttyS1).
> 
> Also both the screen and the touchscreen are controlled over spi so we could even get that functionality back but with Klipper!
> 
> Bonus points: This can all be delivered by hijacking the firmware update process of the adv3 by simply plugging in a USB.

This was my original plan, but I didn't have the time to get it fully working at the time. I'd be happy to collaborate on this if you'd like. I ultimately came to the conclusion that it's just too many wires to solder for *full* functionality with a hardmod, and no one would want to solder such a huge bundle of mod wires, though I did get the screen to partially init using the Pi.

hw-lunemann, 2023-04-29:
If you keep working on the stm32 side of things (I'm not sure what/if anything still needs to be done) then I can take care of the Linux side of things.
Have you had to deal with the uboot setup on the adv3 yet?

VioSynthax, 2023-04-29:
> If you keep working on the stm32 side of things (I'm not sure what/if anything still needs to be done) then I can take care of the Linux side of things. Have you had to deal with the uboot setup on the adv3 yet?

Yes, I ultimately replaced it with a build of U-boot with network recovery https://github.com/widora/u-boot-mt7688

hw-lunemann, 2023-04-29:
Oh the installed version already has that.
I load my image into memory with `tftpboot` already.

Does the widora u-boot work without modifications?
Getting that installed would make it even easier to have both Linux installs in parallel.

ldw300, 2023-04-29:
Wasn't there a similar project like this before? I remember seeing someone on Reddit, maybe on GitHub; that hijacked the firmware update process and flashed a normal Linux install. 

Either way, that sounds like a much cleaner way of doing things. If my main board usb actually worked, (which I'm pretty sure I fried the main board or the Mediatek board when I was poking around too much) I would definitely be interested in helping testing more.

I might get another main board if I find one for a decent price. But right now I'm looking at building an interface board to accept the ribbon cables and use an SKR pick I have to control everything.

VioSynthax, 2023-04-29:
> Oh the installed version already has that. I load my image into memory with `tftpboot` already.
> 
> Does the widora u-boot work without modifications? Getting that installed would make it even easier to have both Linux installs in parallel.

Yeah, but the built in one had fewer features, I believe it's older, and it doesn't have the nice web interface for kernel flashing. From what I remember it works just fine without modification, but I think it had the wrong LED pin set, might be some other minor things but it was plenty for booting, web interface, and UART console

hw-lunemann, 2023-04-29:
Are you sure that was about the adv3? 
I went looking for efforts like taht but couldn't find anything at all, that's why I decided to do it myself.

VioSynthax, 2023-04-29:
If someone has done it, it was probably another similar printer using the Finder-Rush firmware

hw-lunemann, 2023-04-29:
> Yeah, but the built in one had fewer features, I believe it's older, and it doesn't have the nice web interface for kernel flashing. From what I remember it works just fine without modification, but [...]

The LED thing should be configurable in the OpenWRT device tree.

Their u-boot is just a horrible, horrible hack job. 
They don't even use the mtd partitions they have set up. `u-boot-env` just get's ignored, the openwrt 14 build in there doesn't boot due to a corrupted squashfs and so on.
The kernel the adv3 does boot with is just an uImage written to 0x10 in the main memory. I'll just hijack the mtd and put the kernel and squashfs there. That way it'll be easy to revert it to stock if needed.

ldw300, 2023-04-29:
I can't remember or find it right now, I just remember finding two hacky methods regarding firmware hijacking or ssh root shell. That was before I found this project which has been the most promising so far.

VioSynthax, 2023-04-29:
> The LED thing should be configurable in the OpenWRT device tree.
>
> Their u-boot is just a horrible, horrible hack job. They don't even use the mtd partitions they have set up. `u-boot-env` just get's [...]

Good idea. I noticed it complained about corruption absolutely constantly, I figured it was either just mine or they were prone to it. Is it actually a bad master image on the eMMC?

hw-lunemann, 2023-04-29:
Yep.

> Good idea. I noticed it complained about corruption absolutely constantly, I figured it was either just mine or they were prone to it. Is it actually a bad master image on the eMMC?

Yep. The kernel kernel in the mtd partition is from 2017 iirc, so a year older than the one they ended up actually shipping with their hackjob method.
The worst part is that there's no way for a user to recover from a bricked install since they use the live environment to update the system. They aren't using a squashfs, these are the stock kernel args:
```Kernel command line: console=ttyS0,115200 noinitrd root=/dev/mmcblk0p1 rootfstype=ext4 init=/init rw rootwait console=ttyS0,115200n8```

The 115200 baud in the boot loader and linux during the early0 part of the boot console is also a lie btw. It's more like 119500 probably because they set up their build system with the wrong system clock or something like that.

hw-lunemann, 2023-04-29:
I will try to get this done until the 11.05 since I won't be available for a while after that.

penz6, 2023-05-01:
Happy to help if needed, currently working on implementing Octoprint with the stock firmware using the wireless TCP connection, lacks queuing prints but should be possible to at least monitor the print and camera. But it would be much more preferable to run an alternative firmware. Especially since I don't even want to think about what it's phoning home about :(

hw-lunemann, 2023-05-09:
Are you sure octoprint will run on 64mb of ram and a 580Mhz cpu?

penz6, 2023-05-09:
Yeah, ended up just running octoprint on a raspi and connecting over the network using the limited API that it exposes.

hw-lunemann, 2023-05-09:
I wanted to post an update on my work since I won't be able to work on it for a couple weeks.

1. The screen and touchscreen have drivers now and I managed to set up a framebuffer device file for the screen, so drawing to it should be extremely easy now. 
2. The usb mux chip is controlled by a simple GPIO pin and I'm pretty sure auto detection of whether a device is plugged in or not is toggled by another but I just haven't gotten around to testing that one yet. 
3. The webcam also has a driver now and should work aswell.
 
I have a rudimentary install script for my port that also backs up the flash and eMMC so this mod is completely reversible.

There's two things that would be very helpful. Fistly a dump of both the stock eMMC and flash just to make sure everything works without having tinkered with the printer and secondly it would be nice if someone could test the widora uboot to make sure it really does work without problems.

I'll finish it up once I'm back and then we should be pretty much all set!

Swamp-Ig, 2023-07-19:
This level of hacking is a bit above my current skills, but happy to contribute testing on my machine if needed.

jabberjabberjabber, 2023-08-23:
I have an Adventurer 4 (actually a monoprice Voxel Pro flashed to an Adv 4). Is there anything I can do to help with this?

VioSynthax, 2023-12-15:
#1 For replacing stock firmware without hardmods

VioSynthax, 2024-02-27:
> I have an Adventurer 4 (actually a monoprice Voxel Pro flashed to an Adv 4). Is there anything I can do to help with this?

Quite possibly! Adventurer 4 support is on the roadmap, I just want to make sure Adventurer 3 is fully featured and easy to set up before moving my attention fully to the 4. I was considering the 5M as well, but since it has enough RAM for native Klipper, FlashForge went and did it themselves. For initial Adv4 support, we just need to know the pins. for the 3, I just probed around with a multimeter with the board out of the machine to figure out where each trace went, and referenced the STM32 datasheet to find the pin name that corresponds to each component. Those go directly into printer.cfg under the correct section for the component in question. 

We also need to know where the main CPU reset pin is so we can get the Linux system out of the way (until I can get open firmware running on it that plays nice with Klipper) and we have to know where the microcontroller's UART pins end up so Klipper and the Pi can communicate. Eventually this will be done through ethernet for those that don't wish to solder.

Adventurer 4 also uses a weird microcontroller. It's very similar to STM32, but unlike most clones, it doesn't directly support STM32 binaries. It seems to be the same MCU as Adventurer 5M/Pro uses, so at the very least the FlashForge fork of Klipper should work on it or be easily modified to. https://github.com/g992/flashforge-ad5m-5mpro-research/issues/8 https://github.com/FlashforgeOfficial/AD5M_Series_Klipper/tree/main

closed 2024-02-27 by VioSynthax