synthread/Klippventurer #8: thermistor calibration

VioSynthax, 2023-03-22:
Thermistor doesn't seem to read correct values on the higher end with current settings, may need to pick a different thermistor type for printer.cfg or define a custom thermistor.

ldw300, 2023-04-05:
Any luck on the new method? If there is something I can do to help, I can try. My machine is sitting idle until there's at least a alpha or beta to test on this repo. I am monitoring the preview branch atm.

VioSynthax, 2023-04-05:
> Any luck on the new method? If there is something I can do to help, I can try. My machine is sitting idle until there's at least a alpha or beta to test on this repo. I am monitoring the preview branch atm.

Greetings,
The preview branch is ready for use, but once you get to part 3, you have to do the setup manually. Basically just run KIAUH on your Pi, install Klipper, Moonraker, and Fluidd, and copy the cfg to the correct place (so just a normal Klipper setup procedure) then copy the .config to ~/klipper, run `make menuconfig` then save and exit from menuconfig, followed by running `make` 

You have to configure your Pi to free up the UART lines, here are the commands to make that easy:
```
sudo sh -c "echo '# Enable UART port' >> /boot/config.txt"
sudo sh -c "echo 'enable_uart=1' >> /boot/config.txt"
sudo sh -c "echo 'dtoverlay=miniuart-bt' >> /boot/config.txt"
sudo sh -c "echo 'dtoverlay=disable-bt' >> /boot/config.txt"

sudo sed -i 's/console=tty1 //' /boot/cmdline.txt
sudo reboot
```
Once the Pi is back and you can SSH back in, `cd` to `~/klipper` again and run the following command `sudo stm32flash -w ~/klipper.bin -R -i -18,23,18:-18,-23,18 /dev/ttyAMA0` And you should be able to use the printer via the Fluidd web interface immediately after flashing. I'll be here if you need any support! Good luck.

VioSynthax, 2024-02-28:
Thermistor calibration is closer, but filaments still all seem to require 15-20°C higher temps than they should before they print correctly. Will need to calibrate it with an external thermistor.