synthread/Klippventurer #9: Klipper.bin

ldw300, 2023-04-10:
> Any luck on the new method? If there is something I can do to help, I can try. My machine is sitting idle until there's at least a alpha or beta to test on this repo. I am monitoring the preview branch atm.

Greetings,
The preview branch is ready for use, but once you get to part 3, follow these steps.

Run KIAUH on your Pi. Install Klipper, Moonraker, and Fluidd, then copy the .cfg from https://github.com/VioSynthax/Klippventurer-Installer/tree/main/configs to the correct place (so just a normal Klipper setup procedure) and copy the .config to ~/klipper. `cd` to ./klipper and run `make menuconfig` then save and exit from menuconfig, now run `make` 

You have to configure your Pi to enable the serial line. Here are the commands to make that easy:
```
sudo sh -c "echo '# Enable UART port' >> /boot/config.txt"
sudo sh -c "echo 'enable_uart=1' >> /boot/config.txt"
sudo sh -c "echo 'dtoverlay=miniuart-bt' >> /boot/config.txt"
sudo sh -c "echo 'dtoverlay=disable-bt' >> /boot/config.txt"

sudo sed -i 's/console=tty1 //' /boot/cmdline.txt
sudo reboot
```
Once the Pi is back and you can SSH back in, run the following command `sudo stm32flash -w ~/klipper.bin -R -i -18,23,18:-18,-23,18 /dev/ttyAMA0` And you should be able to use the printer via the Fluidd web interface immediately after flashing. I'll be here if you need any support! Good luck.

_Originally posted by @VioSynthax in https://github.com/VioSynthax/Klippventurer/issues/8#issuecomment-1496812296_
            

I could be doing something wrong, but I need a sanity check. I went and installed raspberry pi os lite, then ssh'd into it and run the kiauh script to install klipper, moonraker, and fluidd. Then I transfer the cfg to ~/klipper/config and then transfer the config to ~/klipper. After that I run the make menuconfig, quit, then make. Then to make sure I didn't miss something I transferred and ran the klippventurer.sh script from the installer repo. I've attatched the output. It seems I have missed a step or something is wonky because it says there is no klipper.bin. 
[image]

ldw300, 2023-04-22:
So an update on my side. I was able to get everything mostly going on my side but I used the RPI Imager to image an SD card with MainsailOS with Klipper, moonraker, etc. already installed. Compiled klipper.bin successfully, But when trying to flash the main board I get an error, failed to init device and klippy states it is unable to connect to the mcu. I'm still trying to sort this out myself but if there is any insight on this it would be helpful. Below are some screenshots of what I have encountered. 

It is also very possible I have screwed up my soldering. I'm about to check my work to verify I'm not being an idiot.

[image]
[image]

VioSynthax, 2023-04-24:
Greetings!

My apologies for missing your issue at first, I'm traveling but will return home later this week. 
```
sudo sh -c "echo '# Enable UART port' >> /boot/config.txt"
sudo sh -c "echo 'enable_uart=1' >> /boot/config.txt"
sudo sh -c "echo 'dtoverlay=miniuart-bt' >> /boot/config.txt"
sudo sh -c "echo 'dtoverlay=disable-bt' >> /boot/config.txt"
 
sudo sed -i 's/console=tty1 //' /boot/cmdline.txt
sudo reboot
```
Did you run these commands on MainsailOS? You'll need to run them prior to running
 `sudo stm32flash -w ~/klipper.bin -R -i -18,23,18:-18,-23,18 /dev/ttyAMA0`

The installer repo isn't ready for primetime, just an early script with a lot of debug output that will eventually automate the process, but for now doesn't work.

ldw300, 2023-04-24:
No problem! I am just trying tôprovide input and help where I can. I do remember running those commands but I just re-ran them and tried to flash. Still no go.  When running the flashing command the mainboard and fans switch off twice. Is this normal behavior? Going to verify my soldering in a bit and if it seems okay, I'll reflash an sd card and run kaiuh from scratch and retry the steps.

VioSynthax, 2023-04-26:
> No problem! I am just trying to provide input and help where I can. I do remember running those commands but I just re-ran them and tried to flash. Still no go. When running the flashing command the mainboard and fans switch off twice. Is this normal behavior? Going to verify my soldering in a bit and if it seems okay, I'll reflash an sd card and run kaiuh from scratch and retry the steps.
> 
> Edit: After checking and improving soldering, I realized the DFU wire was on wrong GPIO pin, but after correcting issue, still having the same result. I'm going to re-flash the SD and start from scratch.

have you checked your RST point to make sure it it bridged to GND? I don't think your OS is the concern, reflashing probably won't help. You may want to start a minicom session and verify your UART lines are working by bridging RX and TX together (with local echo off) you can check it by bridging the points and then typing into the terminal, if you get nothing at all when they're apart but you get your input echoed back to you when they're crossed, then the serial UART is working. If it types when apart or doubles the input when together, then you have local echo on. Let me know how that goes. Once I get back home, I can run through the setup again and see if there's any little detail the guide is missing. Oh, and can you take some good clear photos of your installation?

ldw300, 2023-04-27:
So I did end up starting from scratch on the OS, this time I just installed 32bit PI OS, ran KAIU. Installed Klipper, Moonraker, and Fluidd. Ran through the setup process again and got the same results. 

I double checked my RST pin to GND with a multimeter and I have good continuity. I did notice I had some solder bridges when reviewing the photos. (I had removed the Mediatek board previously trying to follow the old method.) So I cleaned up the solder bridges. Then tried a reflash. I got the same errors I have been receiving.

 I run through your request with the UART communication with minicom with local echo off. UART is working properly. I feel like I am missing a simple vital step somewhere and plan to read through everything again. 

I apologize for the lighting quality in these photos, the main lighting in my office is bad, so I tried to make it visible with a flashlight. I can re-take and re-upload if needed. 

For context there is some extra wires in the photos. The short grey wire on bottom is just an extra wire I added to the PRST pad to GND to make sure I had a good bridge since my soldering is terrible. The White wire and the Black wire going to and from the GND on the left side of the board comes from the buck converter and goes to the PI so it should be properly grounded everywhere.

 I tried to use the same color DuPont wires as in your photo for ease of troubleshooting. However I was afraid I was bridging pads with the stranded wires and my poor soldering skills, so I added some solid core wires to make an easier connection to the solder pads and soldered them to the DuPont wires.

Last photo is for GPIO validation to make sure I didn't make another mistake.

[image]
[image]
[image]

VioSynthax, 2023-04-27:
> So I did end up starting from scratch on the OS, this time I just installed 32bit PI OS, ran KAIU. Installed Klipper, Moonraker, and Fluidd. Ran through the setup process again and got the same results. 
> 
> 
> 
> I double checked my RST pin to GND with a multimeter and I have good continuity. I did notice I had some solder bridges when reviewing the photos. (I had removed the Mediatek board previously trying to follow the old method.) So I cleaned up the solder bridges. Then tried a reflash. I got the same errors I have been receiving.
> 
> 
> 
>  I run through your request with the UART communication with minicom with local echo off. UART is working properly. I feel like I am missing a simple vital step somewhere and plan to read through everything again. 
> 
> 
> 
> I did try to flash with minicom open in another terminal, is there supposed to be output to minicom while flashing? Because I have no output.
> 
> 
> 
> I apologize for the lighting quality in these photos, the main lighting in my office is bad, so I tried to make it visible with a flashlight. I can re-take and re-upload if needed. 
> 
> 
> 
> For context there is some extra wires in the photos. The short grey wire on bottom is just an extra wire I added to the PRST pad to GND to make sure I had a good bridge since my soldering is terrible. The White wire and the Black wire going to and from the GND on the left side of the board comes from the buck converter and goes to the PI so it should be properly grounded everywhere.
> 
> 
> 
>  I tried to use the same color DuPont wires as in your photo for ease of troubleshooting. However I was afraid I was bridging pads with the stranded wires and my poor soldering skills, so I added some solid core wires to make an easier connection to the solder pads and soldered them to the DuPont wires.
> 
> 
> 
> Last photo is for GPIO validation to make sure I didn't make another mistake.
> 
> 
> 
> 
> 
> [image]
> 
> [image]
> 
> [image]
> 
> 

Swap TX and RX on the Pi side and give it another try, they appear to be switched around.

ldw300, 2023-04-27:
I have tried swapping them out of curiosity and still got the "failed to init device" message, and no communication to fluidd. 

I removed all the mods and tried to return the ad3 to stock but it seems as if the board has been partially flashed or it is dead because lcd comes on but nothing ever shows up on it and the main board fan isn't running like it usually does.

VioSynthax, 2024-02-28:
Should be fixed after 

> I have tried swapping them out of curiosity and still got the "failed to init device" message, and no communication to fluidd.
>
> I removed all the mods and tried to return the ad3 to stock but it [...]

I think I may finally know what's causing this issue. I hadn't noticed this before, but you have a slightly different microcontroller than every other Adventurer 3 I've seen so far. I know the Adventurer 4 and 5M have this one, but didn't know it ever made it into any other than the Adventurer 3 Pro 2. Is this one also an SZ16 mainboard? The model number is in the top left by the motor controllers. If I can get this working on your Adv3, then I'm confident that I can support the Adventurer 4 and Pro 2 as well.