synthread/Klippventurer #14: Support Adventurer 3 CPU (MediaTek MT7688) U-boot and system image

VioSynthax, 2024-02-29:
Before the functions tied to the Linux side of the board will work, we need some software running on it to communicate with the Pi. This is also a vital prerequisite for #1 so we can do this plug-and-play style and forward the commands through ethernet instead of directly wired serial. There are are two real options here, one of which I don't like. We can:

1. Create a script that just places the binaries we want onto the original rootFS, and replace the init script with our own so we have ultimate control over the firmware. This leaves all FlashForge files in place, and doesn't touch your serial number, MAC address, print time, etc. etc.
2. Replace the old version of U-boot with our own, upstream, modern version with USB recovery and USB booting so the printer is essentially impossible to brick in a way that requires disassembly to fix. This doesn't preclude option 2, we can still boot normal stock firmware from the eMMC, and we can even recover stock firmware without it having to be bootable to do so. As it stands right now, if you can't boot, you can't fix it without some hefty skills. This is all fairly [X1 Plus style](https://www.x1plus.net/). We can then build a new rootFS from scratch with the latest supported version of the Linux kernel, OpenWRT, and all its packages. This makes sure that we have all the latest security updates, and we have ultimate control over what our firmware is doing and how new our build is. We can even USB boot this, so we can leave the eMMC completely untouched and stock, and with our custom U-boot we could recover the stock firmware even if it did get corrupt. 

I feel that option 2 is the clear winner. It keeps the original firmware on the eMMC untouched, and only involves installing an updated version of the existing U-boot bootloader the printer ships with. For those with bricked Adventurer 3s, a secondary installation option to install using a SOIC clip or via the original bootloader's UART will be available to get them back up and running with either stock firmware or our open source firmware.

Unfortunately, while supporting a completely unmodified mainboard is possible, it won't be possible to do without adding some additional horsepower. A Raspberry Pi or some other kind of SBC or print host will be necessary to run Klipper. The CPU onboard the Adventurer 3 could certainly run Klipper, but due to it being written in Python there's absolutely no way Klippy is running on a system with only 64MB of RAM. This goes for the Adventurer 4 as well, just not enough RAM for hosting Klippy. ESPECIALLY with Mainsail and Fluidd and KlipperScreen. What I do think they can handle, is forwarding the command queue to the MCU and running KlipperScreen. I don't know what the experience will be like on the Adventurer 3's tiny screen, but the resolution at least does seem to be supported in KlipperScreen, so we'll just have to wait and see.

VioSynthax, 2024-02-29:
The list of features handled by the onboard processor include the LCD and touchscreen, piezo buzzer, front USB port, filament runout sensor, internal USB (camera) and the ethernet port.

rsnodgrass, 2024-06-26:
Agreed, option 2 sounds like a preferred way forward. I read over how X1 Plus works and it is so nice to leave basically everything with the stock firmware alone so it is easy to revert back (e.g. if you need to sell the A3).

Really look forward to running Klipper on an external RPi without hardware modifying my A3. I would have already done your Klipper mods, but have had way too many other priority projects in line. As fun as breaking out my soldering iron for the A3 would be.

THANK YOU!