synthread/Klippventurer #16: Support FF Creator Pro 2 IDEX ?

StijnMut, 2024-03-04:
Will the flashforge creator pro 2 get support via this project. I’ve been wanting to enable klipper on my machine for a long time but haven’t yet found an accessible way to do so.

The-Synthax, 2024-03-04:
> Will the flashforge creator pro 2 get support via this project. I’ve been wanting to enable klipper on my machine for a long time but haven’t yet found an accessible way to do so.

Eventually. I currently plan on supporting as many of the FlashForge printers as possible, I just need to have one in my possession to reverse engineer it and test. 
Looks like it should be possible to install Klipper via the SD card slot on CP2. LCD definitely won’t work without connecting it to a Pi or making some changes to Klipper, but maybe if the machine has an easy Klipper installer then someone with the skills necessary will come along and write a Klipper module for driving displays.

StijnMut, 2024-03-05:
Sounds absolutely great.  Is there any way i can help with the development of this.   I don’t have much experience with coding but would love to help.  Maybe testing some early builds or something

The-Synthax, 2024-03-17:
> Sounds absolutely great. Is there any way i can help with the development of this. I don’t have much experience with coding but would love to help. Maybe testing some early builds or something

I may be able to make some test builds for CP2 once the Adventurers are printing, thankfully someone has made a pin map so I **might** be able to make some test builds without physically having one of these on hand. It would be very difficult to make a full printer config without having one for testing, so someone else would need to tune the printer if I can't find an affordable used or broken one.

StijnMut, 2024-03-17:
Ive found another person using klipper on the creator pro 2 and he has uploaded a klipper config file however i dont really know how he git it installed.
Maybe the config files will help ? 
Foud these files on his website:

[Klipper CP2 Files from eazrael.zip](https://github.com/synthread/Klippventurer/files/14627134/Klipper.CP2.Files.from.eazrael.zip)

The-Synthax, 2024-03-17:
> Ive found another person using klipper on the creator pro 2 and he has uploaded a klipper config file however i dont really know how he git it installed. Maybe the config files will help ? Foud [...]

Installed using an STLink V2 or a clone it looks like. They're often $8-10 on Amazon and can be used to both recover from a bricked microcontroller or dump the contents of one for analysis. I would recommend making a dump before testing anything so you can restore the bootloader if it becomes corrupt. If you want to test, @ me in the #creator-pro-2 channel on Discord and I'll put together a test build for you. Should be possible to install it in a way that allows for easily returning to stock firmware.

StijnMut, 2024-03-17:
Alright I'll order one on Amazon and will try to dump the original first.   I'll @ you when I get to this point.  I don't have much experience with directly flashing firmware to the controllers itself.

cb-agotchi, 2024-03-24:
https://github.com/cb-agotchi/Flashforge-Creator-Pro-2-Configs

https://youtu.be/mIQjz5eHBBo?si=pjdYA_gWlI6WW3qT

I got the Dockterj fork of Klipper working on my FFCP2. Since then something has changed and it wouldn't resolve the ADS1118 so I added an rp2040 and 3950's for temperature duty.

First link is my confirmed working printer.cfg, macros.cfg and orca profile using a RatRig VCast as the base profile.

Hopefully this is helpful as a pinout reference for you guys to build off of.

eazrael, 2024-05-09:
Hi, if you are still working on this. Complete documentation including pin mapping (mostly) are in my wiki https://wiki.evilazrael.de/en/flashforge-creator-pro-2 . Klipper mods are in the two branches ffcp and ffcp_basic at https://github.com/eazrael/klipper/. Just take what you need.