synthread/Klippventurer #19: Config issues...

3Dgraf64, 2024-03-10:
Hi!  Thank you for your work on this project.  I'm a totally noob in Linux.  That's why I rely on you.
A note on your read.me instruction:
  The command "git clone https://github.com/synthread/Klippventurer.git ~" doesnt work as it is.  Linux gives an error as it tries to overwrite the "pi" directory.  You must remove the "~" at the end of the command.

Next when flashing the controller I got the following response:

http://stm32flash.sourceforge.net/

Using Parser : Raw BINARY
Interface serial_posix: 115200 8E1

GPIO sequence start
 setting gpio 18 to 0... OK
 delay 100000 us
 setting gpio 23 to 1... OK
 delay 100000 us
 setting gpio 18 to 1... OK
GPIO sequence end

Got byte 0x01 instead of ACK
Unexpected reply from device on command 0x02

GPIO sequence start
 setting gpio 18 to 0... OK
 delay 100000 us
 setting gpio 23 to 0... OK
 delay 100000 us
 setting gpio 18 to 1... OK
GPIO sequence end

Then fluid won't connect to the controller.  I verified my wiring more than twice.
Any tip to resolve this?
Is it better to remove the CPU board?

Thanks!
Francis

The-Synthax, 2024-03-11:
the GD32 has been the easiest MCU to work with so far. The only one I know requires a different .config is the Nation branded ones. 

Can you share some photos of your soldering? A clear view of both ends of each wire would be most helpful.  Can you also confirm that your LCD remains off during all of this, with no backlight or splash screen?

3Dgraf64, 2024-03-11:
Yes.  The display stays blank.  The fans too.
[image]
[image]
[image]

The-Synthax, 2024-03-11:
The white screen is making me think that the MediaTek chip is interfering with the UART. Maybe the RST in the diagram I have is insufficient to prevent this? You can try the following jumper instead, which will definitely get the MediaTek system out of the way and we'll know for sure if this is the cause.
[image]

3Dgraf64, 2024-03-11:
Definitely.  The Mediatek seem to interfere. After installing this jumper and sending the flash command many times I get random massages: "Failed to read ACK byte", "Got byte 0x01 instead of ACK Unexpected reply from device on command 0x02", "Failed to init device.".  This time the buzzer make some clicks while executing the command.

The-Synthax, 2024-03-12:
> Definitively. The Mediatek seems to interfere. After installing this jumper and sending the flash command many times I get random massages: "Failed to read ACK byte", "Got byte 0x01 instead of ACK Unexpected reply from device on command 0x02", "Failed to init device.", "GET returns unknown commands (0x79) Failed to read ACK byte". This time the buzzer make some clicks while executing the command.

I was able to replicate this issue and solved it in my case. Could you run the following commands on your Pi and see if it fixes the issue for you as well? 

`sudo sed -e s/ console=serial0,115200//g -i /boot/cmdline.txt`
`sudo stm32flash -w ~/klipper/out/klipper.bin -R -i -18,23,18:-18,-23,18 /dev/ttyAMA0`

3Dgraf64, 2024-03-12:
I copied an pasted the first line and this is the response to the sed command:  "sed: -e expression #1, char 2: unterminated `s' command".  There must be something wrong.

The-Synthax, 2024-03-12:
> I copied an pasted the first line and this is the response to the sed command: "sed: -e expression #1, char 2: unterminated `s' command". There must be something wrong. 

My bad, typo'd that line. try `sudo sed -e s/\ console=serial0,115200//g -i /boot/cmdline.txt` then reboot `sudo reboot` before you use stm32flash.

>Edit: I redid all the procedure ending with the same random message results. Is it possible I bricked my controller?

The controller is definitely "bricked" (no working program is written to the storage), but both the MediaTek board and the Pi flash the entirety of the onboard storage every time it needs an update anyway. A microcontroller like these can never truly be bricked as their bootrom is burned into the silicon. If you remove your Pi and all wiring, the FlashForge firmware would reinstall the original MCU binary and you'd be back to stock.

The-Synthax, 2024-03-12:
if you're still having issues and it won't flash after a few attempts, can you do `cat /boot/cmdline.txt` and `cat /boot/config.txt` and give me the output of those?

3Dgraf64, 2024-03-12:
So.  What I did:  Remove all wires and reboot the print letting it "repairs" itself at least 5 time without success.
Now all wires are back.  Tried again the flash command without success.
First cat command returns:  "console=serial0,115200 console=tty1 root=PARTUUID=eb41f98d-02 rootfstype=ext4 fsck.repair=yes rootwait cfg80211.ieee80211_regdom=CA"
Second cat command returns:  "# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
max_framebuffers=2

# Disable compensation for displays with overscan
disable_overscan=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[all]

[pi4]
# Run as fast as firmware / board allows
arm_boost=1

[all]
enable_uart=1
dtoverlay=miniuart-bt
dtoverlay=disable-bt".

The-Synthax, 2024-03-12:
Looks like the Pi is still outputting its serial console to the UART, so it’s interrupting the flashing process. Try `sudo raspi-config` then interfaces>serial. Turn off console output and turn on serial, then reboot. Hopefully this will do the trick.

3Dgraf64, 2024-03-12:
We did a step!  The flash works!
Now I have to do the config files.  This should be a formality
Thanks again!
How can I donate?

The-Synthax, 2024-03-17:
> We did a step! The flash works! Now I have to do the config files. This should be a formality Thanks again! How can I donate?

Great to hear! Have you had success printing with Klipper yet? Let me know if you need any additional assistance, I'm happy to help 😄

We're waiting to hear back from GitHub to be able to accept direct donations. I really appreciate your willingness to donate, all donations go toward helping us get unsupported printers to port the project to.

3Dgraf64, 2024-03-18:
See yourself!
https://youtube.com/shorts/udvLeUi6mUE?feature=share
https://youtube.com/shorts/5JvzBNBNM6Y?feature=share

The-Synthax, 2024-03-22:
> See yourself! https://youtube.com/shorts/udvLeUi6mUE?feature=share https://youtube.com/shorts/5JvzBNBNM6Y?feature=share I did the E3D mod from u/Br0kenScreen (Reddit). Now I can PID tune the hotend. Both bed and hotend can't be pre-heated simultaneously because of the weakness of the power supply. Still tweaking the Orca's profile. Maybe a linear rail mod for the Y axis in the future...

Looking good! Is that hotend getting you much more flow than the original or are you still bottlenecked by the extruder? 

A linear rail mod would be amazing! Dual linear rails may be possible, maybe with a drilling jig to mount them nice and parallel? That may even free up space underneath for dual Y motors if you really want to push the speeds. 

I'm currently planning to release a Stealthburner / Dragon Burner mount for this printer as well as a bracket that uses the extruder mounting holes for a reverse Bowden tube and BTT U2C. Dual linear rails may be possible, maybe with a drilling jig to mount them nice and parallel? That may even free up space underneath for dual Y motors if you really want to push the speeds. 

Update on the donations situation: We're still waiting to hear back from GitHub about the GitHub Sponsors program, but in the meantime we've opened a Ko-Fi page for anyone who wants to support the project. It's live now on the main repo page or at https://ko-fi.com/synthread ❤️

closed 2024-03-22 by The-Synthax

3Dgraf64, 2024-03-22:
Bottlenecks are the fans.  Only 3010.  Cooling is an issue but better than the orginal toolhead.  A mini Stealthburner or Dragonburner would be great.  Next, the extruder.  I raised the current by 100ma.  Have to test with a CHT nozzle.

Before getting into linear rails I have to do some test printing the same part in each corner of the bed since it's driven from the left side.

ko-fi done!

Thanks again!

The-Synthax, 2024-03-23:
> Bottlenecks are the fans. Only 3010. Cooling is an issue but better than the orginal toolhead. A mini Stealthburner or Dragonburner would be great. Next, the extruder. I raised the current by 100ma. [...]

Did you swap out your extruder spring for a stiffer one or print the spacer? If not I’m impressed you’re managing to need more torque. I thought my motor was skipping originally but upon closer inspection, it was actually the idler, the motor wasn’t losing any steps on mine until I put so much tension on it I was squashing PLA till it looked like linguine.

3Dgraf64, 2024-03-27:
Yes, I installed the spacer.  The motor is skipping.  I realized that the extruder motor is heating enough to melt the filament.  Causing blockages.  Probably the cause of your linguine filament.  I had set the running current to 0.9 but I brought it back to 0.8.  I’m toying with the idea of building a Voron M4 extruder now.

The-Synthax, 2024-03-27:
> Yes, I installed the spacer. The motor is skipping. I realized that the extruder motor is heating enough to melt the filament. Causing blockages. I got linguine too. I had set the running current to 0.9 but I brought it back to 0.8. I’m toying with the idea of building a Voron M4 extruder now.

That'd be a fantastic upgrade! BTW, are you in our Discord server? If you use Discord I'd like to add you to our list of supporters in our server.

3Dgraf64, 2024-03-27:
Yes!  I use it as gravel3170.  Just give me the link.

The-Synthax, 2024-03-27:
> Yes! I use it as gravel3170. Just give me the link.

https://discord.gg/ns2pFdhdMW