synthread/Klippventurer #21: Support for Adventurer 5m/ 5m Pro.

mBlinkii, 2024-03-15:
Hello, will there also be a manual for the Adventurer 5m/ 5m Pro. can you get any help with this? I have a 5m and can help/ test.

The-Synthax, 2024-03-17:
Hey there!

We do plan on supporting the 5M (and Pro) eventually, but the current focus is full feature support for the Adventurer 3. Due to the similarity between these machines and the other Adventurers, once the 3 is fully supported, supporting the others won't take nearly as long.

As for how you can help, watch this repo and/or join our Discord for when we start testing on 5M. I won't be able to afford one for a while to port the project to, but the machine does run a custom fork of Klipper from the factory as far as I know, and there is another project working on bringing fully open firmware to the 5M/Pro here https://github.com/xblax/flashforge_adm5_klipper_mod

mBlinkii, 2024-03-17:
Yes, I already knew about the mod, but the AD5M hardware is too weak for more functions. That's why I wanted to connect an RPI to the hardware.

I think I will follow these instructions and give it a try.
https://github.com/g992/flashforge-ad5m-5mpro-research/blob/main/upgrades/klipper_on_ext_cpu.md

The-Synthax, 2024-03-17:
> Yes, I already knew about the mod, but the AD5M hardware is too weak for more functions. That's why I wanted to connect an RPI to the hardware.
>
> I think I will follow these instructions and give it [...]

These instructions are incredibly similar to the method currently published here for the 3, so should work okay enough on the 5M. The only issue with this method is that you lose the any functions that rely on the CPU, which on the 3 includes the screen, camera, USB, filament runout sensor, and speaker/buzzer. Since the CPU is incredibly similar to the 4, our easy installer for the 4 will work on the 5m with almost no modification. 

I had theorized that the CPU and RAM onboard the 5 couldn't handle much more than basic functionality, but didn't have any solid evidence of that. What limitations are you running into with the onboard Klipper?

mBlinkii, 2024-03-17:
Currently it is still a very early version, KlipperScreen does not work reliably, the camera does not always work. 

For example Obico or Mobileraker. I think that will be too much for the hardware.

The-Synthax, 2024-03-23:
Seeing as it only has 128MB of RAM, I'm really surprised it can handle both Klippy and KlipperScreen at the same time. We're working on a firmware for Adventurer 3 that acts as a server to forward all the necessary interfaces to a Raspberry Pi or other Klippy host computer. Both ARM and MIPS builds of it will release simultaneously, so Adventurer 5M should be able to run it as well as the 3 and 4. It won't have support initially for driving the screen from the Pi, but is planned.

mBlinkii, 2024-03-23:
> Seeing as it only has 128MB of RAM, I'm really surprised it can handle both Klippy and KlipperScreen at the same time. We're working on a firmware for Adventurer 3 that acts as a server to forward [...]

That sounds very good, the internal display does not necessarily have to work. i would probably connect a display to the pi and use it.

mBlinkii, 2024-03-25:
I'm really looking forward to it, I've been waiting with the hardware mod. I will be using a larger display as I find it too small for Klipper Screen. The UI is too small to use it properly. I think for such a small display you would have to develop a custom UI. 

I am ready to test it on the 5M. I don't have any problems with it if something goes wrong. I will know how to help myself.

xblax, 2024-03-25:
> I'd like to get it in public hands ASAP, but we're putting a lot of effort into the installer's reliability, reversibility, and general safety net to try to improve the chances that if something does go wrong, you don't need to be a Linux guru or IT professional to fix the printer's OS, it'll just give you an error and fail gracefully back to stock firmware.

You may want to have look at the install script that I wrote for the other mod mentioned here, in case you also use a chroot based approach:
https://github.com/xblax/flashforge_adm5_klipper_mod/blob/master/device_files/install/flashforge_init.sh
https://github.com/xblax/flashforge_adm5_klipper_mod/blob/master/build_overlay/common/etc/init/S00klipper_mod

It is pretty fail-safe. Only the mod files / chroot are copied to printer and then a single init script is installed the "host" linux system.

In general you might find some of the reverse engineered things, like the functions of the load cell tare mechanism or MCU boot code helpful if you plan to add support for the 5M to your mod.